espotek-org/Labrador #228: EspoTek on Debian 10.13 with Lenovo-t520

MoeMaravilla, 2022-09-30:
Hi,

I Received my ExpoTek/Labrador today and anxiously connected it to my system.
I have downloaded EspoTek_Labrador-2cc0678-x86_64.AppImage to my system.

The device gets recognized, however. I'm getting the following error:
`A USB connection was established, but isochronous communications could not be initialised.
This is usually due to bandwidth limitations on the current USB host and can be fixed by moving to a different port.
Please see https://github.com/EspoTek/Labrador/wiki/Troubleshooting-Guide#usb-connection-issues-other-platforms`

I have no other USB devices connected.

When I run `sudo usb-devices` I get the following block for EspoTek/Labrador:
`T:  Bus=02 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=03eb ProdID=ba94 Rev=02.00
S:  Manufacturer=EspoTek
S:  Product=Labrador
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=usbfs
`

Please let me know what I can try next.
Thank you,
Moe

EspoTek, 2022-10-04:
Hi Moe.  That message usually means that your system is lacking the USB bandwidth to support the device.

I don't know exactly why this is the case (it could be that an internal webcam is directly wired to the root hub, or something like that), but the easiest solution would be to connect the Labrador via a USB hub.  Even a cheapie from AliExpress should do the trick, as long as it supports High Speed (480MB/s) or higher speeds.

MoeMaravilla, 2022-10-13:
Does it require a USB-3 port?

EspoTek, 2022-10-13:
Nah, any old USB 2.0 hub should work.