espotek-org/Labrador #231: Calibration has been abandoned due to out-of-range values.

dev-null2019, 2022-11-08:
Hi!
Compiled the software on Linux(Kali)/ARM64.
Sadly Iam not able to calibrate the hardware - the scope shows a signal at 48V on both channels without anything connected.

On my Windows machine the hardware works without problems.

Regards
Stefan

[image]

Logs:

```
└─$ labrador                                                                                       
QObject::connect: No such slot bufferControl::signalGenIn(bool)
QObject::connect:  (sender name:   'signalGenGroup_CH2')
QObject::connect:  (receiver name: 'bufferDisplay')
QObject::connect: No such slot isoDriver::setWindow(int)
QObject::connect:  (sender name:   'timeBaseSlider')
QObject::connect:  (receiver name: 'controller_iso')
QObject::connect: No such slot timedTickBox::resetTimer(bool)
QObject::connect:  (sender name:   'lockPsuCheckBox')
QObject::connect:  (receiver name: 'lockPsuCheckBox')
Attempting to open "/usr/bin/EspoTek-Labrador/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/usr/bin/EspoTek-Labrador/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Attempting to open "/usr/bin/EspoTek-Labrador/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/usr/bin/EspoTek-Labrador/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
driver =  unixUsbDriver(0xaaaaf8edd8f0)
axes =  QCustomPlot(0xaaaaf898a9f0, name="scopeAxes")
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
Device found!!
KERNEL DRIVER INACTIVE
Interface claimed!
Connected
BOARD IS RUNNING FIRMWARE VERSION 0x0006
EXPECTING FIRMWARE VERSION 0x0006
FIRMWARE VARIANT = 0x02
EXPECTED VARIANT = 0x02
Connecting now!
newDig
isoCtx submitted successfully!
[n, k] =  0 0
t =  1667901958987
Delay =  0
Setup successful!
MAIN THREAD ID 0xffffa6db3020
Iso Stack initialised!
SUB THREAD ID 0xffff849cefe0
PSU DUTY CYCLE of dutyTemp = 16 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
PSU DUTY CYCLE of dutyTemp = 11 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
PSU DUTY CYCLE of dutyTemp = 6 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
PSU DUTY CYCLE of dutyTemp = 1 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
PSU DUTY CYCLE of dutyTemp = 0 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
unified debug is of size 27
dt_userWantsToCalibrate 16384
QObject::connect: No such slot MainWindow::enableLabradorDebugging()
QObject::connect:  (receiver name: 'MainWindow')
ReinitUsb Stage 1 complete
Low resolution detected: 1280 x 371
Cancelled 0 0
Total Cancelled = 1
libusb_cancel_transfer FAILED
ERROR LIBUSB_ERROR_NOT_FOUND
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 459, resource id: 11573832, major code: 40 (TranslateCoords), minor code: 0
Cancelled 1 0
Total Cancelled = 2
libusb_cancel_transfer FAILED
ERROR LIBUSB_ERROR_NOT_FOUND
Cancelled 2 0
Total Cancelled = 3
libusb_cancel_transfer FAILED
ERROR LIBUSB_ERROR_NOT_FOUND
Cancelled 3 0
Total Cancelled = 4
libusb_cancel_transfer FAILED
ERROR LIBUSB_ERROR_NOT_FOUND
Cleaning... #3 phases remain.

Cleaning... #2 phases remain.

Cleaning... #1 phases remain.

Cleaning... #0 phases remain.

Cleanup complete
ReinitUsb entering stage 2

unixUsbDriver destructor ran!
isRunning? false
THREAD Gone!
Transfers freed.
Interface released
Device Closed
Libusb exited
unixUsbDriver destructor completed!

genericUsbDriver dectructor entering
genericUsbDriver dectructor completed
Reinitialising USB driver!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
ReinitUsbStage2 is returning
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
Device found!!
KERNEL DRIVER INACTIVE
Interface claimed!
Connected
BOARD IS RUNNING FIRMWARE VERSION 0x0006
EXPECTING FIRMWARE VERSION 0x0006
FIRMWARE VARIANT = 0x02
EXPECTED VARIANT = 0x02
Connecting now!
deviceConnectedDisplay::connectedStatusChanged running!
newDig
isoCtx submitted successfully!
[n, k] =  0 0
t =  1667901962254
Delay =  0
Setup successful!
MAIN THREAD ID 0xffffa6db3020
Iso Stack initialised!
New voltage = 4.5
SUB THREAD ID 0xffff849cefe0
Going to send value  31
newDig
Trigger Level:  -4
Trigger sensitivity: 1
Trigger Level:  -4
Trigger sensitivity: 1
Trigger Level:  -4
Trigger sensitivity: 1
Trigger Type:  0
Trigger Type:  0
Trigger Type:  0
Changed to mode 0
newGain = 8.000000
gainMask = c0c
Multiplying by  0.0625
Buffer shifted by -4
unified debug is of size 27
Changing workspace...
scopeIn_CH2 true
Trigger Level:  -71
Trigger sensitivity: 1
Trigger Level:  -71
Trigger sensitivity: 1
Trigger Level:  -71
Trigger sensitivity: 1
Trigger Type:  0
Trigger Type:  0
Trigger Type:  0
Changed to mode 2
Calibration routine beginning!
unified debug is of size 27
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 741, resource id: 11575540, major code: 40 (TranslateCoords), minor code: 0
unified debug is of size 27
unified debug is of size 27
unified debug is of size 27
VRef (CH1) =  7.45155
VRef (CH2) =  7.40503
unified debug is of size 27
unified debug is of size 27
```

EspoTek, 2022-11-09:
Not sure how it was compiled, but something clearly went wrong!

I'd try again with `-fsigned-char`.

mi-hol, 2025-02-10:
@dev-null2019 is this still an open issue?