espotek-org/Labrador #232: Oscilloscope range clipping

MrDerples, 2022-12-02:
The oscilloscope display will clip voltages past a certain level until channel 1 is paused and unpaused. The negative range seems to be clipped at -1.37 V and/or -4.65 V, while the positive channel seems to clip at +4.65 V. I've observed this behavior on both channels under a number of different operating conditions, and the steps below seem to be the bare minimum needed to replicate it:

1) Open the Labrador software. Connect a signal source between oscilloscope channel 1 (DC) and ground, and feed it a signal that will exceed -1.37 V and/or +4.65 V. For purposes of illustration, I used a 1 kHz, 10 Vpp sine wave centered on zero, but the nature of the signal, or whether or not the Labrador's signal generator is being used to provide it, doesn't seem to matter. I also set the trigger level to zero, but you can observe this with an untriggered acquisition, too. Note the channel 1 max and min values.

[image]

2) Change the vertical range to something that should show the whole waveform. In my example, -6 V to +6 V. You will observe that the waveform is clipped at the levels previously mentioned, like in the image below.

[image]

3) Pause acquisition (click the "Paused" checkbox"), then un-pause it - the waveform will no longer be clipped.

[image]

For reference, I'm running on Windows 10 Enterprise 21H2, build 19044.2311. The Labrador software is git hash 2cc0678, downloaded from Github on 2022-12-02 (New York time). I've been able to replicate this on multiple Labrador boards.

EspoTek, 2022-12-03:
Very clear report.  I know exactly what would cause this and will get it fixed up after Christmas.

mi-hol, 2025-02-10:
@MrDerples is this still an open issue?

mi-hol, 2026-02-15:
@mmehari would you perhaps be in a position to triage this issue?

mmehari, 2026-02-15:
There has been a lot of changes in the code since the issue was first reported and I couldn't replicate it anymore. Perhaps @EspoTek or @MrDerples can talk about its status

mi-hol, 2026-02-15:
@mmehari thanks so much for trying, my assumption was that it would have been fixed already

closed 2026-02-15 by mi-hol