espotek-org/Labrador #233: Oscilloscope trigger stops working upon changing vertical range

MrDerples, 2022-12-02:
When using the oscilloscope trigger, the displayed waveform will stop updating if, first, the vertical range is expanded below -1.14 V or above +4.54 V, and then, the oscilloscope is paused and un-paused. I noticed this because pausing and un-pausing the scope is how I work around issue #232.) This only seems to happen once per program session, the first time either of those range values is exceeded, and any of the following seems to resolve it:

- Toggling any oscilloscope channel on and off;
- Toggling the trigger on and off;
- Jogging the trigger value up and back down;
- Entering the trigger level textbox and pressing enter, re-entering the currently-displayed value.

It's almost like, when the range is changed past the thresholds mentioned, the trigger level gets changed within the program, but without updating the display.

For reference, I'm running on Windows 10 Enterprise 21H2, build 19044.2311. The Labrador software is git hash 2cc0678, downloaded from Github on 2022-12-02 (New York time). I've been able to replicate this on multiple Labrador boards.

EspoTek, 2022-12-03:
>It's almost like, when the range is changed past the thresholds mentioned, the trigger level gets changed within the program

It does!  Past those voltage thresholds, it'll update the PGA and therefore the map to convert digital samples to analog values (and vice-versa).  I'll look into why this isn't updating in the post-Christmas break, and make sure to never again use retained mode GUIs for as long as I live!

MrDerples, 2023-01-07:
Do you have an ETA for fixes for issues #232 and #233? I ask not to be 
pushy, but because I'm planning to use Labradors in a course, and I'm 
deciding whether or not to include the ritual for clearing these bugs in 
the relevant lesson. We're probably not going to start using them until 
February 8th though, so there's some time.

Thanks,
- MrDerples
-- 
Nicholas Johnson, Ph.D.
Instructor
Virginia Tech Department of Biomedical Engineering and Mechanics

EspoTek, 2023-01-09:
Hi mate.  I don't think they'll be fixed up soon.  All of the post-Christmas downtime I had was eaten up by a COVID infection.