espotek-org/Labrador #237: logic analyzer - I2C decoder

abondis, 2023-02-28:
I'm having an issue with the logic analyzer in I2C mode
it seems that I'm able to see the WRITE commands in the serial decoding window, but for some reason it gets cleared very fast. 
Is there a way to:
1. keep everything in this decoding window?
2. freeze, for example capture X seconds and freeze then analyze later?

EspoTek, 2023-03-01:
Hi mate.  As far as I'm aware, the serial decoding screen should literally never clear, unless you're sending a whole lot of newlines.

Is it formatted as hex or ASCII?

tonyn79, 2024-01-29:
Almost a year old, but I am having trouble with the i2c analyzer. I have a esp outputting to an oled and every once in a while I can see the WRITE: followed by a handful 0x00. But it is only 1 or 2 lines of that, then nothing else. Has anyone got i2c to decode successfully?

EspoTek, 2024-01-31:
What frequency is it running at?  100k should be fine, 1MHz won't work.

tonyn79, 2024-01-31:
I have tried at 100 and 400. I can see the wave forms in logic analyzer but no dice so far on serial output. I confirmed with another usb analyzer and it is outputting the packets. My use of the ESPOTEK is logic analyzer inputs (1 and 2) going to sda, scl. Both LA turned on with serial decoding.  Every once in a while I see WRITE: with a number of 0x00, but thing consistent coming through. Any ideas to what I may be doing wrong?

EspoTek, 2024-01-31:
This might sound silly, but have you tried swapping the SDA and SCL inputs around?  If not, there might be some sort of odd regression in the software.

abondis, 2025-02-10:
Hi, I haven't had the opportunity to use it for i2c since then. So I can reopen an issue later if I happen to encounter it again. Not sure for the other person on the thread, I'll let them answer.