espotek-org/Labrador #243: Labrador Development Documentation

damercer, 2023-04-20:
Hello:
I am a software developer. I am trying to adapt my laboratory virtual instruments software package to use the Labrador hardware. After reading through the "Labrador Development Documentation" file here in your repository I'm having some difficulties. It is apparent that the documentation is currently very out of date. I guess my first question would be is there any possibility of seeing new USB interface support documentation anytime soon? My development language of choice is Python and the pyusb interface library.

Using what info there is and a lot of guessing and sniffing the USB traffic to the board using Wireshark I have managed to be able do some communication with the board and control some parts of the hardware such as toggle the digital output pins and adjust the PSU voltage. I can send waveform arrays to the awg output channels but nothing appears on the output pins. That is from a cold power up of the board. If I first run your provided software and configure the awg outputs to something and close the software, I can then send awg data using my test program ( dev.ctrl_transfer() ) to the board and see it on the outputs. But only the part that happens to be the same length (number of samples) as what was used in the Labrador software. Clearly I'm missing something here that starts the output stream.

On the input side, I can send control settings to set the scope channel mode and gain with out any errors but when I attempt to read back data using dev.read(0x83, readarray) as mentioned in the documentation all I get is an error saying:
"NotImplementedError: Operation not supported or unimplemented on this platform" So again I seem to be missing something here as well and the output stream needs to be initialized and/or started beforehand.

Any help on sorting this out would be greatly appreciated. As background on my laboratory virtual instruments software package I have adapted versions that currently work with three other USB based data acquisition platforms and is currently being used by hundreds of colleges and universities around North America and the world. The Labrador board would be number four.

Thanks.

EspoTek, 2023-04-23:
Hi Doug.  No plans to write new documentation, as it's unlikely to be useful to many people.

Feel free to share your code, either publicly or via email and I'll be more than happy to provide support directly.

Cheers,
~Chris

mi-hol, 2026-02-15:
related to #371