espotek-org/Labrador #247: Device not Detected Windows

hjrjacquez, 2023-05-07:
Good day,

I was working on a project and midway through it the software stopped recognizing the board but the light still turns on. My laptop runs on Windows 10, it's a Lenovo Ideapad.
I tried the following:
Restarting the computer,
Getting a new cable (It's one that can transfer data)
Uninstalling and reinstalling,
Tried looking for the device in the device manager as stated in the guide but no go.
Tried on another laptop with a fresh install and still did not recognize the board.

When I plug the debug console I get the following:
[image]

Does anyone know how to solve this?

EspoTek, 2023-05-07:
It sounds like the board popped while you were working on your project.

Could the voltage have spiked above 20V on any of the pins?  Did you smell any magic blue smoke when it stopped working?

hjrjacquez, 2023-05-08:
Thank you so much for your response. I was working with 5V. No smell or smoke came from it. Is there any way I can test if it popped? This is the last time it worked, I was making this measurement. 
[image]

EspoTek, 2023-05-08:
Weird.  Please contact me at admin@espotek.com and I'll organise a free replacement for you.

If you're not getting any "USB Connected" noise at all, the board is cooked an there's nothing you can do.

Cheers,
~Chris

closed 2023-05-08 by EspoTek