espotek-org/Labrador #249: Unable find App on Google Play

ralphxyz, 2023-06-05:
I cannot find the Labrador App on Google Play.
Ralph

EspoTek, 2023-06-05:
Should be right here:
https://play.google.com/store/apps/details?id=org.qtproject.example.Labrador

If you can't install it via Google Play, please get back to me with your phone model, Android version and country of residence so I can look into it.

ralphxyz, 2023-06-05:
Thanks Chris, I could not find it using my phone.
I was able to follow your link and have it installed!
I got a message about the App being developed on an older version but it
installed.
Now I have to find my board.

Ralph

geiseri, 2023-09-23:
I cannot install it on ChromeOS because it is too old.  Would it be possible to update the Play Store version?  It says it was last updated in 2018.

deltaroe, 2023-10-22:
Ran into the same issue myself in the US with a Pixel7 running Android 14 (arm64-v8a arch) 

Made an attempt to get the android build working myself and got decently far, but ran into an issue with the fftw3 library being missing and I don't see there being an android version referenced so not sure what I'm doing wrong.

EspoTek, 2023-10-22:
The APK for Android is available here:
https://github.com/EspoTek/Labrador/releases/tag/android

I don't know why Google stopped showing it on certain phones, but I can look into it.

deltaroe, 2023-10-22:
That apk says it's not compatible with my phone and I haven't figured out
why yet.

geiseri, 2023-10-27:
I think it was using an older NDK. I can't get it to build at all so I am not sure how to fix it. There seems to be a binary library that it can't find.

EspoTek, 2023-10-31:
Man, there has really been a spike in interest for this app on Android.

I'll set aside some time before the end of the year to get it ported across again.

n2ballroy, 2024-03-23:
Same problem here.  Google states the app is not compatible with pixel 7 pro running android 14

EspoTek, 2024-03-25:
Noted, I'll set this as priority #1 for the next dev day!

Digarton, 2024-06-03:
My apologies to bother, is any news here about new release since this one from last year: https://github.com/EspoTek/Labrador/releases/download/android/Labrador-Signed.apk
Still doesn't accept and not installing on Samsung S21 with Android 14.
;-)

EspoTek, 2024-06-04:
> My apologies to bother, is any news here about new release since this one from last year: https://github.com/EspoTek/Labrador/releases/download/android/Labrador-Signed.apk Still doesn't accept and not installing on Samsung S21 with Android 14. ;-)

No news... yet!

phedders, 2024-09-05:
It would be so helpful to be able to run this on an android tablet....

mi-hol, 2025-02-10:
[image]

turboencabulator, 2025-05-09:
The latest [continuous build](https://github.com/espotek-org/Labrador/releases/tag/continuous) may fix the compatibility warning.  The minimum and target SDK versions are 21 and 28, respectively, which correspond to Android 5.0 "Lollipop" and 9.0 "Pie".  I don't get a warning on Android 10.  Please help us test and let us know if there are still Android versions that are incompatible.

As for the version in the Google Play store, I suspect we will need to make the app AArch64-compatible to get it submitted to Google Play.

EspoTek, 2025-05-09:
I get the warning on my Pixel 5 running Android 14.

Apparently it's caused when your target SDK version is at least 2 versions older than the Android running on your device (for example, my Android 14 device is happy with apps that target either 33 or 34):
https://developers.google.com/android/play-protect/warning-dev-guidance (see section "Android App Compatibility Too Low Warning")

That said, apparently we need to target SDK level 34 at a minimum to get onto Google Play:
https://developer.android.com/google/play/requirements/target-sdk

Can we just bump the number at it works OK?  My main concern is that we're doing some sneaky stuff to get an isochronous libusb stream.

mi-hol, 2025-05-09:
> The minimum and target SDK versions are 21 and 28, respectively, which correspond to Android 5.0 "Lollipop" and 9.0 "Pie".

Could there be a build for 4.4?
Many tables are not upgradeable to later versions, this would give them a new life for use as Labrador display

EspoTek, 2025-05-09:
Apparently 5.12 can support SDK versions as low as 16 (Android 4.1).

However 5.15 requires 21 (Android 5.0) or later.

turboencabulator, 2025-05-20:
I've just pushed a change to build with Qt 5.12 instead of 5.15.  Unfortunately for me, it's still too new since all my tablets run Android 4.0.3 or 4.0.4 (SDK 15).

Target SDK is still at 28 and I don't understand all the implications of changing it.

russdill, 2025-05-27:
Would love to buy one of these, but a working android app is a must for me. Very frustrating that Google does this to small developers. Hopefully it can get resolved.

mi-hol, 2025-05-27:
@russdill a working Android app is directly downloadable from the project's GitHub releases [here](https://github.com/espotek-org/Labrador/releases)

[image]

mi-hol, 2025-05-27:
@russdill forgot to mention that "expand Assets" is required to see it

[image]