espotek-org/Labrador #251: [FR]

slepiavka, 2023-07-03:
My Labrador arrived yesterday - congratulations on bringing this amazing project to life. It's insane  how a tiny, cheap PCB can have better support and documentation than many full-blown commercial solutions.

Anyways. I saw an Amazon question regarding higher input voltages, where you suggested to `[...] Just solder on an additional resistor in parallel with R3 and R4 and then rebuild the software [...]`. I can mess around with thorugh-hole, but this circuit is **TINY**! How would you suggest for a user to go about this? Not sure where a resistor would even physically fit (assuming I could even SMD solder).

Maybe I'm missing something: is there a drawback to just using a good old voltage divider in the kOhms range? I don't speak signal processing, but I understand that dividers can introduce some capacitance/inductance? I'm not dealing with high frequencies or anything fancy.

EspoTek, 2023-07-04:
Hi mate.  Don't be imtimidated by SMD soldering!  It takes a bit of skill to do it with an iron, but it's dead simple with paste.

Just get yourself a syringe of cheap solder paste (keep it refrigerated), a knockoff 858D from AliExpress and some bent-nose pliers.  You'll be able to swap out little 0603 components like the resistors with no effort.  Just make sure you don't set the fan speed too high, or you'll start blowing the other components away and make a great big mess.

Or, as you mentioned, you can use a voltage divider in the kOhms range.  The inductance/capacitance wouldn't make a difference at the low frequencies Labrador can measure, but the additional resistive load could change things!  Also, make sure you use resistors with enough wattage to handle the higher voltages you're passing through them.  And please, don't kill yourself by being reckless with 120V/240V AC...

slepiavka, 2023-07-04:
Thank yo so much. To be clear... When you said "additional resistor in parallel with R3 and R4", do you mean literally squishing another resstor right next to R3 sharing the leads and another one on R4? [image]

EspoTek, 2023-07-04:
I have no idea what I meant when I wrote that comment 3 years ago, but either method should be OK.

photonicsfox, 2023-07-12:
Scaling v down is easy with resistive dividers for Vin above design voltages.  Do be aware that depending on resistor type, the resistive divider's impedance may not be linear with frequency, esp once going into the realm > 100kHz.

mi-hol, 2025-02-10:
@photonicsfox @slepiavka would you mind to capture above learnings into a new wiki page https://github.com/espotek-org/Labrador/wiki
Maybe  "Enhance hardware capabilities" would be a good title?