espotek-org/Labrador #253: Couple Hardware questions

MATRIX7878, 2023-09-18:
Hello @EspoTek,

     I got the board as I needed an Oscilloscope and power supply and I have a couple questions:

1. For the power supply, the max is 12VDC.  Do I need to plug it into a separate power source?  Or can I leave it in my USB connection to my laptop?
2. Also, for the power supply, the jumper connection, which side is on and which is off?
3. Is there going to be an update to also measure inductance?  I have some inductors and do not know what values they are as there are no markings.

Thank you,

Drew

EspoTek, 2023-09-19:
Hi mate.

To answer your questions directly:

1) Don't connect an external power supply to the board.  It has a boost converter that generates up to 12V from the USB's power.

2) The side that makes the little red LED turn on is the on position.  :)

3) No plans to add inductance measurements, unfortunately.  I remember looking into it and without modifying the hardware the precision was nowhere near good enough to measure things in the mH range.

Enjoy the board!

closed 2023-09-19 by EspoTek

MATRIX7878, 2023-09-19:
Hello,

     Thank you for letting me know.  I appreciate it.  Sad about the inductance, but understandable.

Drew