espotek-org/Labrador #256: Noise on pin 42 (o'scope ch. 2) due to signal on pin 40 (ch. 1)

dbsamithey, 2023-11-16:
I noticed that noise appears on ch. 2 of the oscilloscope when a signal with frequency >~ 1 Khz and peak-to-peak magnitude >~ 2 V is applied to ch. 1.  As shown in the gif below, it strangely goes away when the ch. 1 signal has a peak-to-peak magnitude  >~3.2 V. This threshold appears to be insensitive to the PSU setting.  Also, the noise appears irrespective of whether the DC or AC pins of the signal generator and oscilloscope are used.
The same type of noise doesn't seem to show up on ch. 1 under any conditions.  On the other hand, if the hardware pins assigned to ch. 1 and ch. 2 are swapped in the board firmware, the noise starts to appear on ch. 1 instead of ch. 2.  I'm surprised about this hardware pin specificity because the input stage on the schematic looks identical for the two pins.  Please let me know if you have an idea as to how the noise might be mitigated.
[image]

EspoTek, 2023-11-17:
This is bizarre.

I know there's a "Signal Gen Triple" setting that pumps the signal gen output through an amplifier when >3.2V, but if anything this should reduce the noise when <3.2V.  Do you see the same artifacting when on a period that's not an integer multiple/divisor of 1ms?  (e.g. 4691Hz)?

dbsamithey, 2023-11-20:
I still see the problem at that frequency.  Also, the threshold for the onset of the noise does look to be exactly 3.2 V, so the "Signal Gen Triple" setting seems relevant.  But, if the noise were purely related to the signal generation, then I would maybe expect it to show up regardless of whether the signal generator is plugged into channel 1.  In reality, the noise goes away when the signal gen is disconnected even when the output setting is kept at some large voltage ~< 3.2 V.
This suggests that the noise could be related to the measurement of the large waveform.  I checked this by analyzing the smaller signal in the default 1 channel mode of the Labrador while the larger signal was applied to the unsampled oscilloscope input.  The noise only ever appears on the pin 42 input, while the default 1 channel mode shows data from the pin 40 input, so for this test I again flashed a modified firmware with the hardware pins assigned such that the pin 42 input was shown in the 1 channel mode.  Then, with pin 42 measuring a ~400 mV waveform and a ~3.1 V waveform applied to pin 40, again the noise appeared.  So, the presence of a large signal on pin 40 is sufficient for the noise to appear, and actual measurement of the waveform is irrelevant.
Maybe the noise goes away when the voltage setting of the large signal is > 3.2 V because then the chip puts out a signal that is three times smaller.  The noise does in fact appear to have the same level when the voltage setting of the signal applied to pin 40 is 3.21 V / 3 = 1.066 V as when it is 3.21 V.  As far as I understand, for both these settings, the peak-to-peak magnitude of the signal produced by the chip is 1.066 V, so this observation is consistent with the chip-generated voltage being relevant.  If that's true though, then it's all the more confusing that actual application of the signal to pin 40 is important.  The increment 3.19->3.21 V only substantially changes the conditions on the chip's signal gen output pin, so it seems like it should not be decisive whether the signal is applied to pin 40.

The screenshots below I think clarify things a bit.  One is with a wire connected between the ch. 1 (pin 40) input and some piece of metal that's picking up 60 hz mains hum and with nothing connected to the ch. 2 (pin 42) input.  This results in strong distortion of the ch. 2 input.  On the other hand, with ch. 2 instead picking up mains hum and with nothing connected to ch. 1, no strong distortion shows up on ch. 1.  It seems like, generally speaking, the ch. 2 input is more susceptible to distortion when there's a signal on ch. 1 than vice versa, and the max voltage amplitude on the chip's signal gen pins is just one variable that controls the level of the distortion.

Ch. 1 picking up mains hum; Ch. 2 not connected to anything
[image]

[image]
Ch. 2 picking up mains hum; Ch. 1 not connected to anything

mi-hol, 2025-02-10:
@dbsamithey is this still an open issue?