espotek-org/Labrador #259: Enabling Logic Analyzer CH1 (mode 1) disables Signal Gen CH2

mmehari, 2024-03-02:
I had the impression that Signal Generators (CH1 and CH2) run continuously, since they have dedicated DMA channels and they don't  require USB ISO traffic, but I found out that when enabling Logic Analyzer CH1 (mode 1), Signal Gen CH2 gets turned OFF.

Furthermore, I looked inside the firmware code where the DMA channel is set (https://github.com/EspoTek/Labrador/blob/master/AVR_Code/USB_BULK_TEST/src/tiny_dma.c#L139-L155) and indeed the DMA source address is set to dummy but not to Signal Gen CH2.

Is there any reason why this is the case?

EspoTek, 2024-03-03:
It's been almost a decade since I wrote that code (oh my gosh, I'm old!), but there's some niggling feeling at the back of my mind that it's absolutely not an accident.

I _think_ it has something to do with the way that the digital inputs are serialised (IIRC it pumps garbage data into a SPI configured as MOSI, just to create a clock signal for the two other SPIs which are configured as MISO).

This might be a hangover from an old version of the code, though.  Do you have the tools or knowhow to compile a new software version and program the board?

mmehari, 2024-03-05:
I still haven't figured out how Signal Gen CH2 is affected by the SPI but indeed it is not by accident.

I also saw your comment on https://github.com/EspoTek/Labrador/blob/master/Desktop_Interface/ui_elements/buffercontrol.cpp#L86
> //Signal Gen CH2 doesn't work with bus sniffer 

Although I can compile and upload a new firmware, I will keep it "as it" for now.

closed 2024-03-08 by EspoTek