espotek-org/Labrador #260: Extending Labrador vendor requests

mmehari, 2024-03-14:
Hello again,

This time I am planning to add a new feature which requires a new vendor request but while experimenting to add one, I ended up having the following error message

> unixUsbDriver::usbSendControl FAILED with error LIBUSB_ERROR_PIPE

What I did to recreate the error message was updated SIGNAL GEN CH1 command from **0xa1** to **0xb1** in the following two files
1. https://github.com/EspoTek/Labrador/blob/master/AVR_Code/USB_BULK_TEST/src/ASF/common/services/usb/udc/udc.c#L1108
2. https://github.com/EspoTek/Labrador/blob/master/Desktop_Interface/genericusbdriver.cpp#L235

Would you please let me know what I am missing?

EspoTek, 2024-03-15:
Hi again, Michael!

At least in theory, that should have done the trick!

In practice, though, there's a feature that automatically flashes the "correct" firmware to the board that could be interfering with your code.  To rule this out, it'd be best practice to increment the `EXPECTED_FIRMWARE_VERSION` in the desktop app, and the `FIRMWARE_VERSION_ID` in globals.h on the microcontroller side.

If you're running on any operating system that isn't Windows x64, you'll also want to uncomment the line defining `SINGLE_ENDPOINT_INTERFACE` (also in globals.h).  This changes the headers so that it uses 1x 1023-byte isochronous endpoint, rather than 6x 128-byte endpoints to send the scope/logic analyzer data.  On non-Windows-x64, the ordering of the transfers isn't deterministic, so using 6x endpoints leads to a corrupt signal on the oscilloscope/logic analyzer.

The other thing I'll mention in general is that Wireshark is an excellent tool for capturing raw(ish) USB packet data.  You can see exactly what data is sent to the device, and exactly what the response is.  Just make sure you check the "install USBPcap" box when installing it if you're on Windows (don't install USBPcap separately - it doesn't work properly).

mmehari, 2024-03-15:
Indeed you were right.

My firmware was overwritten by the default firmware and increasing the FIRMWARE_VERSION_ID solves the issue.

And yes I am already using Wireshark to debug the low level USB traffic and indeed it is very helpful.

Now I will focus on the new feature I am trying to bring and hopefully, I will see you in a new pull request

EspoTek, 2024-03-16:
Great news!  Can't wait to see the PR!  I'll mark this issue as closed now.

closed 2024-03-16 by EspoTek