espotek-org/Labrador #261: Unable to download software to raspberry pi

lcarletti88, 2024-03-18:
I am trying to download the labrador software to a raspberry pi, I've followed the instructions for the raspberry pi build, which is just to copy the following command into a terminal

wget -O /tmp/labrador_bootstrap_pi https://raw.githubusercontent.com/EspoTek/Labrador/master/labrador_bootstrap_pi && sudo chmod +x /tmp/labrador_bootstrap_pi && sudo /tmp/labrador_bootstrap_pi

However when I run into the following error

make: g++-8.1.0: No such file or directory
make: *** [Makefile:793: buffercontrol.o] Error 127

It seems make is unable to find the compiler in the location specified by $PATH. I've ensured the compiler named "g++-8.1.0"  does exist in the place described by $PATH with no typos. I am unsure where to go from here.

Link to raspberry pi bootstrap code for convenience: https://raw.githubusercontent.com/EspoTek/Labrador/master/labrador_bootstrap_pi

Thank you in advance

EspoTek, 2024-03-18:
This is a little bit concerning.  Are you able to provide a full terminal dump?

The following command will generate the dump and save it to "espotek_log.txt" in your home directory:
`wget -O /tmp/labrador_bootstrap_pi https://raw.githubusercontent.com/EspoTek/Labrador/master/labrador_bootstrap_pi && sudo chmod +x /tmp/labrador_bootstrap_pi && sudo /tmp/labrador_bootstrap_pi > ~/espotek_log.txt`

lcarletti88, 2024-03-19:
Thanks for the response, I've attached the dump.

[espotek_log.txt](https://github.com/EspoTek/Labrador/files/14652856/espotek_log.txt)

EspoTek, 2024-03-19:
Sorry, my bad!  The log command I gave earlier logs messages but not errors.  Extremely un-useful!

Correct command is:
`wget -O /tmp/labrador_bootstrap_pi https://raw.githubusercontent.com/EspoTek/Labrador/master/labrador_bootstrap_pi && sudo chmod +x /tmp/labrador_bootstrap_pi && sudo /tmp/labrador_bootstrap_pi > ~/espotek_log.txt 2>&1`

lcarletti88, 2024-03-19:
No worries, here's the new file.

[espotek_log.txt](https://github.com/EspoTek/Labrador/files/14656611/espotek_log.txt)

EspoTek, 2024-03-19:
Bizarre.

It looks like you have a partial install of the gcc version that Labrador uses.  I don't know how that happened.
Try running the following command then trying again:
`sudo rm -rf /usr/local/gcc-8.1.0-labrador`

lcarletti88, 2024-03-20:
I'm still getting the same error, I've attached the dump although it looks mostly the same barring it acknowledges the gcc is not already installed. I may try to factory reset the pi tomorrow and try again.

[espotek_log2.txt](https://github.com/EspoTek/Labrador/files/14668720/espotek_log2.txt)

EspoTek, 2024-03-21:
I've received an email from another customer who has mentioned their journey getting it to run.  Are you on the 64-bit version of Raspbian?  That's currently unsupported, and I should probably get around to that...

DeeJay, 2024-03-21:
This is 'the other customer'.

Note the emphasised text in the README under the Raspberry Pi heading: "Please note that the 32-bit version of Raspbian version 9 (Stretch) or later is required to install this software."

My recent experience is that with the 32-bit (armhf) version of the current Bookworm [Debian 12] release of the supported Raspberry Pi Operating System [RasPiOS] the supplied labrador-bootstrap-pi script runs to completion without error, producing and installing a runnable version of the Labrador app.

lcarletti88, 2024-03-21:
Thank you both. I was using a 64 but OS. I misinterpreted "or later" as including a newer 64 bit version would also be fine, I'm not an expert by any means with this stuff.

Appreciate the help

mi-hol, 2025-01-12:
@EspoTek may I suggest to rename the title to "Unable to compile software to raspberry pi 64bit OS"

mi-hol, 2025-01-12:
duplicate of https://github.com/EspoTek/Labrador/issues/219
#219

closed 2025-02-10 by mi-hol