espotek-org/Labrador #262: Device Not Recognized on Windows

jtleniger, 2024-03-20:
Hello! I just received my new Labrador and I'm unable to get it recognized on my Windows machine.

I've tried 3 different USB cables, at least one of which is known to work, as well as every 2.0 and 3.0 port on the machine with everything but my mouse unplugged.

I've tried reboots and reinstalling the Labrador software including all 3 included drivers.

I don't see anything in the device manager labeled libusbK or Labrador in any category.

I hope it wasn't a DOA board, but I'm not sure what else to try. Perhaps another machine or OS?

EspoTek, 2024-03-20:
Does it make the "USB Connected" noise when you plug it in?

jtleniger, 2024-03-20:
Chris, I'm a proper idiot and mixed up my dozens of microUSB cables and only realized it when I retested with a known good microUSB device.

I apologize. I've found the surefire way for me to solve problems is prematurely opening issues, cheers :)

closed 2024-03-20 by jtleniger

EspoTek, 2024-03-20:
Haha, no worries.  You'd be surprised how many problems I've solved just by starting to type out a forum post then closing the tab...