espotek-org/Labrador #264: Introducing duty cycle to the Square waveform

mmehari, 2024-03-24:
Hi again,

For a project I am working on, I want to use a square waveform with configurable duty cycle.

The current version only has fixed (50%) duty cycle and I want to make this configurable.

I am planning to add a **Duty cycle** item underneath Signal Gen CH1/2 front-end but regarding the back-end implementation, I am thinking of creating the 'duty-cycled' Square waveform inside **sendFunctionGenData** function (https://github.com/EspoTek/Labrador/blob/master/Desktop_Interface/genericusbdriver.cpp#L144) all from scratch .

This, however, will effectively disregard the waveform data (***Square.tlw**) stored locally and I wonder if this is the right approach.

Please let me know if you have other ideas.

Thanks,

EspoTek, 2024-03-24:
Go for your life, mate.  I think that's the best way to approach it, even if it's a little bit inconsistent.

Have seen your other PR too and will respond during the week!

mmehari, 2024-03-30:
Well I couldn't find another approach and once the other PR is finalized, I will issue a new PR for this.

mmehari, 2025-01-13:
Since the serial encoding PR is finalized, I have created a new PR on this issue. https://github.com/EspoTek/Labrador/pull/270

mmehari, 2025-02-15:
PR is finalized. Closing the issue.

closed 2025-02-15 by mmehari