espotek-org/Labrador #266: Is my dog dead?

OldSurferDude, 2024-06-24:
When I grounded a _probe_ is saw that it was below zero so I thought to recalibrate.  I removed all of the devices connected and started the calibration.  Several times I got the out of range error message.  And then, the not connected message.  The light was on.

I tried all the usual things and noticed that the Espotek was no longer showing up in Device Managers (Windows).  Usual things being: uninstall/re-install; reboot; power cycle; restore restore point.  No luck.

Is my dog dead?

Thanks for your consideration.

EspoTek, 2024-06-25:
Could be dead, could just be sleeping.  :)

There's a PTC fuse on the board that protects it from some, but not all shorts.  If you leave it unplugged for a minute or two that usually resets the fuse.

Cheers,
~Chris

OldSurferDude, 2024-06-25:
"Despite the valiant efforts of the nerd doctors, the dog valiantly clings to death."  (apologies to _That Was the Week That Was_ reference to Francisco Franco)

closed 2024-06-25 by OldSurferDude