espotek-org/Labrador #267: Capacitor measurement

MATRIX7878, 2024-08-14:
Hello,

     I want to be able to measure my capacitor value and I see that I need a reference resistor to try to get the RC timing down to 1 ms.  I do not know the capacitor value, that is what I am trying to figure out.  Am I missing something when it comes to the multimeter capabilities for capacitors?

Thank you

EspoTek, 2024-08-14:
Hi mate, there's a little bit of guesswork involved with this but you don't need to be too precise.  As long as the RC constant is somewhere in the order of 1-10ish ms, the software can work out the rest.

If you're using a ceramic cap, try 10k or 100k.  For an electrolytic, try 100 ohm or 1k.
If the waveform you see is stupidly long, drop the resistance.  If it's too short, increase it.

Again, you don't need to be exactly 1ms, but if you're way too short you'll run into quantisation errors and if you're way too large you'll run into nonlinearities.

MATRIX7878, 2024-08-14:
Hello,

     Alright.  I understand.

Thank you

closed 2024-08-14 by MATRIX7878