espotek-org/Labrador #268: Add GUI option to work better on smaller screens

j-lucas-d, 2024-12-30:
Hi,

As you can see in the attached picture that the Labrador GUI does not fit on this 7" monitor. Would it be possible to add an option to modify the GUI, so that it fits better on smaller screens? Perhaps something like a tabbed layout for all sections (display on one, oscilloscope controls on another, generator on another, etc). I think this would help greatly with utilizing available screen real estate.

Thank you for a terrific little product which augments my custom rPi computer.

[image]

EspoTek, 2025-01-12:
3 or 4 people have given this feedback, especially for the 720x480 displays.

It's really something I should do!

mi-hol, 2025-01-12:
duplicate of #248

closed 2025-02-10 by mi-hol