espotek-org/Labrador #289: Spectrum display should make use of windowing

tdunning, 2025-02-14:
When doing a spectral analysis, it is important to multiply each window of input samples by a so-called windowing function. This causes samples at the beginning and end of the window being reduced to zero with the original amplitude being retained in the middle of the window.

The reason that this is important is that the Fourier transform assumes the input is periodic. A batch of samples isn't, however, and there can be considerable mismatch between the beginning of the sample window and the end. That mismatch causes very poor quality in the quality of the display.

As an example, here are a few cycles of a sine wave:

[image]

You can see how the first sample on the left is 0 and the last sample is about -1. If you do an analysis that presumes this sampled signal repeats these samples over and over, there will be a big jump from -1 back to 0 when you go from the end back to the beginning.

It is better to use a windowing function. A good choice is $2 \thinspace \sin^2({\pi t} / (n \Delta t))$. This changes our sine wave into this:

[image]

Here, the windowing function is shown as a light red ghost on top of the windowed signal. 

In the spectral display, this makes a lot of difference because the step from the end back to the beginning is an impulse that raises the noise floor dramatically.

For instance, here is a plot of the signal above with and without windowing. The difference in noise floor is about 150dB which is massive.

[image]

In such a spectral displace, the vertical display should be calibrated using a factor based on the number of samples in a window and the gain of the input signal. If we want to display dBm (dB referenced to 1mW), and we have n = 10000 samples in our data and the input gives us a value of 127 for a 1V input, a 1dBm signal would appear to have a magnitude of 109dB after doing the Fourier transform which is computed using

$$s_{ref} = 20 \thinspace \log_{10} (127 \thinspace v_{0dB} \thinspace n)$$

where $v_{0dB} = \sqrt{0.05} = 0.22$ is the voltage that results in 1dB into a $50 \Omega$.

EspoTek, 2025-02-18:
Wow, that makes a lot of sense.  Thanks for the super clear explanation!

tdunning, 2025-02-18:
Happy to help as much as I can. Which, sadly, isn't much.

tdunning, 2025-02-20:
Yes. There are lots of possible alternative windows. Having a selection can be nice.

On the other hand, there really is not a large difference between different windowing algorithms. As such, a simpler user experience might be nice.

mmehari, 2025-02-20:
Indeed a simpler interface is better for the user experience and I am thinking of having a single drop-down list.
When the user picks a window from the list, he/she will see the windowing effect on the spectrum

tdunning, 2025-02-20:
Since I am not doing the implementation, I am very conscious that my
recommendation should carry nearly as much weight as a real implementor.

A good default is good enough for me.

Having a pull down that doesn't steal real estate from a more important
choice is a fine addition.

Having you do the work to make it all happen dominates all of these other
considerations. All hail the implementor!

mmehari, 2025-03-24:
We can close this issue since related PR is now merged

closed 2025-03-24 by mmehari