espotek-org/Labrador #302: Linux AppImage missing udev rules

not-reed, 2025-03-16:
AppImage seems to not be working for me. at first I thought none of my USB cables had data, however after testing on my mac, it worked flawlessly.

when starting the AppImage the logs begin with
```
Installing udev rules to /run/udev/rules.d/69-labrador.rules
/tmp/labrador-udev-helper0HdjV0: line 3: /run/udev/rules.d/69-labrador.rules: No such file or directory
```

then after starting it gets stuck on a loop

```
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/028, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
DEVICE NOT FOUND
```

and the app never detects the device

turboencabulator, 2025-03-16:
Do you have a `/run/udev/rules.d` directory?  Maybe this script needs to create it first before writing files there.

mi-hol, 2025-03-16:
@not-reed please provide Linux version

not-reed, 2025-03-16:
I'm using https://endeavouros.com/

I don't have the directory, but I do have `/etc/udev/rules.d` which I have added rules to in the past. It seems this is where they live on Arch https://wiki.archlinux.org/title/Udev

I tried making `/run/udev/rules.d`, and copying the file here: https://github.com/espotek-org/Labrador/blob/master/Desktop_Interface/rules.d/69-labrador.rules which solved the `Installing udev rules to /run/udev/rules.d/69-labrador.rules` error, but it still complained about permissions. Unfortunately ran out of time today before I could solve, but thought I would open an issue here.

turboencabulator, 2025-03-17:
Thanks for the feedback!  I've updated the AppImage to create `/run/udev/rules.d` if it doesn't exist.  We used to put this file in `/etc/udev/rules.d`, but `/run/udev/rules.d` seems like a more appropriate spot for the AppImage to install it to, since it's volatile.  As long as udev can find the file, either location should be fine.

If it's still not setting up the permissions properly, my guesses are:
1. The udev tools don't know to look in `/run/udev/rules.d` for config files.  This seems unlikely, I was just browsing the systemd sources and it has been looking here for at least the past 10 years.
2. There is an extra command you need to run after installing the file:  `udevadm control --reload-rules && udevadm trigger --subsystem-match=usb`.  This will reload all config files and update the permissions if your device is currently plugged in.  The AppImage should run this command for you, but only if it doesn't find the `69-labrador.rules` file installed anywhere.

Can you try the following and let us know if it's fixed?
1. Delete any `69-labrador.rules` files you've manually installed.
2. Reboot, to make sure all of the old rules are gone.  I expect that your `/run/udev/rules.d` directory will be deleted over the reboot.
3. Try the latest AppImage (version 4a713e0).

not-reed, 2025-03-19:
I appreciate the work!

Just downloading and running the AppImage didn't work at first.

But running it directly from the command line prompted for my sudo password (makes sense reading the script) and after that it worked.

Not sure if that's just an issue on my end missing the system prompt. From a quick search it appears pkexec doesn't pass along env variables on Wayland so these need to be added `pkexec env WAYLAND_DISPLAY="$XDG_RUNTIME_DIR/$WAYLAND_DISPLAY" XDG_RUNTIME_DIR=/run/user/0` however I haven't tested this yet, maybe there are better solutions, or the problem is entirely on my end like i said (possible)

not-reed, 2025-03-23:
reporting back, my polkit auth agent wasn't running properly. works correctly now 🤝 thanks!

closed 2025-03-23 by not-reed