espotek-org/Labrador #310: Actions: Build Workflows for Windows and Android failing

mi-hol, 2025-03-30:
@espotek-org/maintainers could someone please look into the affected build workflows and fix them?

[Windows](https://github.com/espotek-org/Labrador/actions/runs/14155303839)
[Android](https://github.com/espotek-org/Labrador/actions/runs/14155312687)

mi-hol, 2025-03-30:
> Android

@turboencabulator could below be ralted to https://github.com/espotek-org/Labrador/pull/305 ?
[image]

mi-hol, 2025-03-30:
> Windows

@EspoTek Chris, could the root cause be a missing credential for espotek-org?
in line:
**Add-Content "$env:USERPROFILE\.git-credentials" "https://$env:ACCESS_TOKEN:x-oauth-basic@github.com`n"**

```

 # 8) Optional: Push result to Labrador-win-builder
      - name: Publish update to Labrador-win-builder
        if: success()
        shell: pwsh
        env:
          ACCESS_TOKEN: ${{ secrets.ACCESS_TOKEN }}
          OUTPUT_BIN_DIR: ${{ matrix.OUTPUT_BIN_DIR }}
        run: |
          mkdir C:\dev
          cd C:\dev
          git clone --depth 1 https://github.com/espotek-org/Labrador-win-builder
          cd Labrador-win-builder
          git config --global credential.helper store
          Add-Content "$env:USERPROFILE\.git-credentials" "https://$env:ACCESS_TOKEN:x-oauth-basic@github.com`n"
          git config --global user.email "admin@espotek.com"
          git config --global user.name "Chris Esposito"
      
```

turboencabulator, 2025-03-30:
> [@turboencabulator](https://github.com/turboencabulator) could below be related to [#305](https://github.com/espotek-org/Labrador/pull/305) ?

No, this is because we've never added the fftw dependency to any of the Android code.  We have neither the library to link to, nor the header that it's trying to include here.  Same goes for openmp and eigen, see my recent comment in #282.

turboencabulator, 2025-03-30:
> [@EspoTek](https://github.com/EspoTek) Chris, could the root cause be a missing credential for espotek-org?

Multiple problems here:
1. The URL is wrong, that repo is still in `EspoTek`, not `espotek-org`.
2. If you know where to look, that credential is public in an old config file.  Hopefully that credential has been revoked so not just anyone can commit anything to that repo.
3. Even if it was in `espotek-org`, there are too many manual steps.  It's committing the build artifacts to a different repo, then it requires a separate action to go build the windows installer from the contents of that other repo, then another action to take the results and add them back to the release in this repo.

mi-hol, 2025-03-30:
> 3\. Even if it was in `espotek-org`, there are too many manual steps.

@turboencabulator would you have a suggestion how a better setup would look like (i.e. reference to another project building for multiple OS platforms)?

turboencabulator, 2025-03-30:
The current Windows workflow builds the executable and gathers all the necessary files for it to run (I think, I haven't tested it).  It's just missing the steps to combine everything into a Windows installer.  I haven't found time yet to set up a Windows machine or to take those parts of the `Labrador-win-builder` repo and add them to this workflow.

mi-hol, 2025-04-02:
> 1. Advanced Installer.  I've added a step to register Advanced Installer if the `AI_KEY` secret is present, but I don't have a key nor permission to set up the secret.

@EspoTek Chris, would you have the required key?

mi-hol, 2025-04-02:
> Multiple problems

@turboencabulator there is an additional issue regarding:

4. update version id with every release automatically

currently it seems to stay at 1.0.0

[image]

EspoTek, 2025-04-03:
> Update on the Windows build: I've copied over the build steps from the Labrador-win-builder repo and the job is passing. Please test the [artifacts](https://github.com/espotek-org/Labrador/actions/runs/14210676095).
> 
> Known issues:
> 
> 1. The installer and uninstaller both have a nag screen that it was built with a trial version of Advanced Installer.  I've added a step to register Advanced Installer if the `AI_KEY` secret is present, but I don't have a key nor permission to set up the secret.
> 2. The version of Advanced Installer it's using (15.7) has rules to install the VC++ 2017 redistributable, but not any later versions.  We need at least the VC++ 2019 redistributable to match what Labrador was built with.  I've set up the job to copy the 2019 .exe over the 2017 .exe it's expecting.  This is an ugly hack, the installer probably won't update to 2019 if 2017 is present.  The solution is to either roll back to MSVC 2017 (painful to set up), update Advanced Installer, or switch to a different installer builder program.
> 3. The installer prompts to install the drivers every time, even if they're already installed.
> 4. I think it would be good to split the 32-bit and 64-bit installers instead of having a combined one.  That way it doesn't install an extra 30-45 MB of files that won't be used, and it would simplify the Github job.
> 
> Once 2 is fixed and we've tested it, I'll set up the job to update the release.

Issues noted.
I'll contact the company and get a new key, which should fix 1.
For 2, simplest way is to fix the .ai files manually.  I've done it before, it's not that painful if you know what you're doing.  I'll put it in my notes to get done.
For 3, it's not ideal but I don't think it's a killer.  People aren't going to reinstall often, and if they do they can always just skip (or spend 20 seconds extra clicking through the driver install).
For 4, I think it's best for the 32 and 64 bit versions to remain unified on Windows.  Many users aren't aware of architecture differences, and an extra 30MB download size is preferable to confusing people.
For 5 (version not incrementing), I suspect that'll just be a command line thing too.  I'll check with the company and see if that's easy to set at build time.

EspoTek, 2025-04-15:
I've made a new branch that should fix all of the issues with the Windows build, but unfortunately it looks like I'm unable to test it right now as download.qt.io is down.

Branch is here: https://github.com/espotek-org/Labrador/tree/testing-new-installer

mi-hol, 2025-04-15:
> 5\. update version id with every release automatically
> 
> currently it seems to stay at 1.0.0

seems fixed via showing github commit ID

[image]

mi-hol, 2025-04-15:
@EspoTek have you had any luck with the other planned actions?

turboencabulator, 2025-04-16:
> [@EspoTek](https://github.com/EspoTek) [download of MSFT C-runtime](https://github.com/espotek-org/Labrador/commit/43122537f35184d4abdd7d4ea522353a1748b54e#diff-5ce5c9ff86f58b1f87b35b5227b2e84cb69f022d6741e1854f3e1e181091773cR33-R84) is currently commented out. Did you encounter issues with that part?

This looks like a bug.  It's still grabbing the same `vc_redist` files that `windeployqt` finds, rather than the ones that Advanced Installer wants to use.

@EspoTek I've pushed a branch (`vcredist-download`) with the download URLs that Advanced Installer gives.  Can you reset the `master` branch to here?  Doing that will get rid of ~30MB of `vcredist` files in the git history (that we weren't using), and from what I've read, Microsoft doesn't want anyone else to host these files for various reasons.  I believe that branch will work, but I'm also hitting the same Qt outage so I can't really test it.

turboencabulator, 2025-04-16:
> > [@EspoTek](https://github.com/EspoTek) I've pushed a branch (`vcredist-download`) with the download URLs that Advanced Installer gives. Can you reset the `master` branch to here? Doing that will get rid of ~30MB of `vcredist` files in the git history (that we weren't using), and from what I've read, Microsoft doesn't want anyone else to host these files for various reasons. I believe that branch will work, but I'm also hitting the same Qt outage so I can't really test it.
> 
> Have pulled it across to `master`. I doubt it'll cause any issues.

I was hoping to rewind/delete commit 43122537f35184d4abdd7d4ea522353a1748b54e so `git clone` and such won't have to download these large files...

mi-hol, 2025-04-16:
> Yep, got all of them done except for 3 (The installer prompts to install the drivers every time,

When I tried yesterday there was still "1. The installer and uninstaller both have a nag screen that it was built with a trial version of Advanced Installer."

EspoTek, 2025-04-16:
> > Yep, got all of them done except for 3 (The installer prompts to install the drivers every time,
> 
> When I tried yesterday there was still "1. The installer and uninstaller both have a nag screen that it was built with a trial version of Advanced Installer."

https://github.com/espotek-org/Labrador/issues/310#issuecomment-2808799032

That one there should no longer have the nag.  At least whatever is running on my local machine is good.

> > > [@EspoTek](https://github.com/EspoTek) I've pushed a branch (`vcredist-download`) with the download URLs that Advanced Installer gives. Can you reset the `master` branch to here? Doing that will get rid of ~30MB of `vcredist` files in the git history (that we weren't using), and from what I've read, Microsoft doesn't want anyone else to host these files for various reasons. I believe that branch will work, but I'm also hitting the same Qt outage so I can't really test it.
> > 
> > 
> > Have pulled it across to `master`. I doubt it'll cause any issues.
> 
> I was hoping to rewind/delete commit [4312253](https://github.com/espotek-org/Labrador/commit/43122537f35184d4abdd7d4ea522353a1748b54e) so `git clone` and such won't have to download these large files...

Done.

mi-hol, 2025-04-16:
@EspoTek  Did you see [Invalid workflow file: .github/workflows/windows.yml#L83](https://github.com/espotek-org/Labrador/actions/runs/14485459478/workflow)
You have an error in your yaml syntax on line 83 ?

EspoTek, 2025-04-17:
> [@EspoTek](https://github.com/EspoTek) Did you see [Invalid workflow file: .github/workflows/windows.yml#L83](https://github.com/espotek-org/Labrador/actions/runs/14485459478/workflow) You have an error in your yaml syntax on line 83 ?

Yep, already sorted.

mi-hol, 2025-04-17:
> but I'm also hitting the same Qt outage so I can't really test it.

Windows Build action fails with below error but qt download website is reachable.
Did they maybe change something permanently (i.e. mirror sites)?

https://github.com/espotek-org/Labrador/actions/runs/14514118752/job/40719240081#step:4:180

```
INFO    : Finished installation of opengl32sw-32-mesa_12_0_rc2.7z in 1.88799010
WARNING : Caught ChecksumDownloadFailure, terminating installer workers
ERROR   : Failed to download checksum for the file '5.15.2-0-202011130602qtactiveqt-Windows-Windows_10-MSVC2019-Windows-Windows_10-X86.7z' from mirrors '['https://download.qt.io/']
==============================Suggested follow-up:==============================
* Check your internet connection
* Consider modifying `requests.max_retries_to_retrieve_hash` in settings.ini
* Consider modifying `mirrors.trusted_mirrors` in settings.ini (see https://aqtinstall.readthedocs.io/en/stable/configuration.html#configuration)

```
[image]

mi-hol, 2025-04-17:
update:
there seems to be a workaround in https://github.com/jurplel/install-qt-action/issues/285#issuecomment-2812447172

mi-hol, 2025-04-17:
@espotek-org/maintainers Windows build is working again
https://github.com/espotek-org/Labrador/actions/runs/14524341888

mi-hol, 2025-04-18:
> That one there should no longer have the nag. At least whatever is running on my local machine is good.

I can confirm Windows installer from latest "continuous build" solved that issue

mi-hol, 2025-04-23:
@turboencabulator Android build works now! Great work again!

I'll close this issue now and create new issues from the comments in PR #323

closed 2025-04-23 by mi-hol

turboencabulator, 2025-05-09:
> When I tried yesterday there was still "1. The installer and uninstaller both have a nag screen that it was built with a trial version of Advanced Installer."

I just discovered that Github secrets weren't being passed from one workflow to another.  That might explain why you still got the nag screen, it wasn't actually registering.  Should be fixed now.