espotek-org/Labrador #312: Logic Analyzer: Voltage values inconsistencies

iansmith1112, 2025-03-31:
It would be nice to have logic analyzer voltage values be represented accurately and scale with V_min and V_max. 

Images show that waveform does not scale or represent correct voltage values.

[image]

[image]

mmehari, 2025-03-31:
Can you describe the setup so that we can recreate the issue?

iansmith1112, 2025-04-02:
> Can you describe the setup so that we can recreate the issue?

Setup: Windows 11 application analyzing I2C waveform at 100kHz

mmehari, 2025-04-02:
What I meant was to know how you generated the I2C waveform. How did you setup your hardware?

EspoTek, 2025-04-03:
Unfortunately this isn't possible due to the way the hardware is set up.

The logic analyzer pins are wired to the gate of a MOSFET, so there's no way to tell whether the input voltage is 3.3V, 5V or 12V.

Going to close this one off for now, sorry mate!

closed 2025-04-03 by EspoTek