espotek-org/Labrador #313: Logic Analyzer: Ch1 and Ch2 Voltage offset

iansmith1112, 2025-03-31:
It would be nice to be able to offset Ch1 and Ch2 of the logic analyzer to see both waveforms not overlayed one another. 

Image shows that both channels are overlayed and cannot be shifted, making it hard to compare data and clock lines. 

[image]

mmehari, 2025-03-31:
Can you describe the setup so that we can recreate the issue?

iansmith1112, 2025-04-02:
> Can you describe the setup so that we can recreate the issue?

Setup: Windows 11 application analyzing I2C waveform at 100kHz

mmehari, 2025-04-02:
What I meant was to know how you generated the I2C waveform. How did you setup your hardware?

EspoTek, 2025-04-03:
It looks like he's just hooked up the SDA and SCL lines to the Logic Analyzer CH1 and CH2 pins on the bottom-right:
https://github.com/espotek-org/Labrador/wiki/Pinout

I totally agree that an offset would be nice here!

iansmith1112, 2025-04-04:
Sorry for the delayed response. The image is a snapshot of a single 100kHz I2C communication with the on-board gyroscope of a STM32F072B MCU. You are right that I was trying to analyze the SDA and SCL lines but wasn't able to offset them to analyze them. 

My work around was using the export to csv feature, but having an offset would be very nice.

EspoTek, 2025-04-05:
> Sorry for the delayed response. The image is a snapshot of a single 100kHz I2C communication with the on-board gyroscope of a STM32F072B MCU. You are right that I was trying to analyze the SDA and SCL lines but wasn't able to offset them to analyze them.
> 
> My work around was to use the export to csv feature, but having an offset would be very nice.
> 
> To add on: Yes, I am using the header on the bottom right of the board for the LA's channel 1 & 2 and connecting them to the SDA and SCL lines of my STM32 MCU.

If you click the "serial decoding" button, it should decode automatically to a little text box below the scope display.  You might need to swap around the channels that SDA and SCL are connected to, though.