espotek-org/Labrador #314: Issue with AppImage use on Raspberry Pi 5

emergentia, 2025-03-31:
Apologies if this has been addressed elsewhere, but I am still having issues with getting things running on the RPi 5 on the 64-bit ARM. I was hoping that the new release here:

https://github.com/espotek-org/Labrador/releases/tag/continuous-Linux

would address the challenge. Getting it up on MacOS has been easy enough. The RPi5 is running on Debian GNU/Linux 12 release 12 (bookworm), so I used the AppImage per the instructions. Executing the AppImage simply produces a text file with no content and random characters in the title. The .deb package install throws a 'Failed to install file' error. 

Has anyone gotten the Labrador software to run on the RPi5?

Thanks!

emergentia, 2025-04-01:
> The continuous-Linux release is meant for x86_64 architecture but RPi 5
> has aarch64 architecture and it will not work by default. I guess we need
> to have a separate release for aarch64 architecture.
>
> For now, you can compile it yourself but not on the current master branch.
> I have issued a pull request for making the necessary changes but as a
> quick fix, you can pick up from the PR #315
> <https://github.com/espotek-org/Labrador/pull/315> and issue the
> following commands.
>
> cd Desktop_Interface/
> mkdir build
> cd build
> qmake QT_ARCH=aarch64 ../
> make -j4
> sudo make install

Thank you, Michael. I will try that! On another note, I had the Labrador
running on my Mac yesterday. I did take using the Firmware Recovery
function but worked fine. Later in the day, I started to see the Device Not
Connected! indicator pop up, and the recovery method no longer works. I
don't have any other peripherals connected to the Mac via USB per the
instructions. Is there something else that I need to do? I did follow some
guidance on using Homebrew to update the libusb. Do you recommend a
manual wire of the firmware as described here:

https://github-wiki-see.page/m/espotek-org/Labrador/wiki/Troubleshooting-Guide

Any help is appreciated.

Best,
Adam

mmehari, 2025-04-02:
Hi Adam,
As much as I would like to help you, I don't have Mac to try it out and it is best if others give comment on this one.

mi-hol, 2025-04-02:
> pull request for making the necessary changes

@emergentia PR was merged into "master"  please provide feedback

emergentia, 2025-04-02:
Understood on the Mac topic. The issue I am seeing indicates that there is
something going on at the micro USB connector. When the board was first
received, the connect was a bit bent and wouldn't allow the cable to be
inserted. Gently prying the connector a bit allowed the cable to be
inserted, but the red status light flickered unless the cable was jiggled
into a specific position. That was followed by a few successful connections
using the firmware recovery, but once unplugged/re-plugged I have had no
success. It seems like the power is stable at this point. Just seems like
there is an issue with the data connection. I ordered a new cable (USB-C to
micro USB) to avoid using a dongle. If that doesn't work, I will
likely just order a new Labrador board.

-Adam

emergentia, 2025-04-02:
Re: the aarch64, I tried to build yesterday using the instruction above but still ran into issues with the libdfuprog during the make step.

mmehari, 2025-04-02:
I forgot to mention about a new commit I did to clarify the ambiguity of identifier and architecture naming https://github.com/espotek-org/Labrador/commit/b6937f69e40c5bccc7769ffa2f760f820a2b1868.

These are the commands you should run now

```
cd Desktop_Interface/
mkdir build
cd build
qmake QT_ARCH=arm64 ../
make -j4
sudo make install
```

mi-hol, 2025-04-02:
> Re: the aarch64, I tried to build yesterday

@emergentia no local build required any longer due to merge in our master branch!
Just download from https://github.com/espotek-org/Labrador/actions/runs/14225869384

[image]

turboencabulator, 2025-04-03:
> [@emergentia](https://github.com/emergentia) no local build required any longer due to merge in our master branch! Just download from https://github.com/espotek-org/Labrador/actions/runs/14225869384

That build doesn't have the `aarch64` image yet, but the current one does.  You can now find an `aarch64` AppImage on the [releases](https://github.com/espotek-org/Labrador/releases) page.

emergentia, 2025-04-03:
The new AppImage build works on my Pi5, and Labrador is running. Still the same issue with the connection. I'll just order a new board. Thank you for all the help!

turboencabulator, 2025-04-03:
Good to hear it works!  Let us know if you have any issues with that AppImage build.

closed 2025-04-03 by turboencabulator

EspoTek, 2025-04-03:
Wow, how did I miss this?  Does it work on 32-bit as well, or just ARM64?

emergentia, 2025-04-03:
Not sure as I only have a Pi5. Runs from the AppImage direct and via terminal. Thanks again!

turboencabulator, 2025-04-03:
> Wow, how did I miss this? Does it work on 32-bit as well, or just ARM64?

Right now we only have a job to build the AppImage on 64-bit ARM.  32-bit ARM would also work if Github offered a 32-bit ARM runner.