espotek-org/Labrador #322: APPIMG & Deb failed in Linux | Source failed in OpenBSD

anoduck, 2025-04-18:
Two computers:

1. Kali Linux Rolling Release (Debian Sid) 
3. OpenBSD Current Snapshot (OpenBSD 7.7)

## OpenBSD Desktop Build

_Specs_:
- Clang: 16.6
- Qmake-qt5: 3.1
- QT: 5.15.16
- GCC 8.4.0

Attempted to build from source on OpenBSD with Clang, received error that `omp.h` could not be located. Understanding that `omp.h` is optional for Clang, moved to attempting to build with Gcc. Build with Gcc failed with error:

```c++
desktop_settings.cpp:26:16: error: aggregate 'timeval tv' has incomplete type and cannot be defined
 struct timeval tv;
```

Abandoned build on OpenBSD.

## Headless Kali Linux Install

Next tried to install on Kali Linux:

### Install with Deb

Began with `.deb` package to save system space. Received error regarding unmet dependency, `qt5-default`. 

```bash
Unpacking espotek-labrador (2.0) over (2.0) ...
dpkg: dependency problems prevent configuration of espotek-labrador:
 espotek-labrador depends on qt5-default; however:
  Package qt5-default is not installed.

dpkg: error processing package espotek-labrador (--install):
```

The problem is, there is no installation candidate for qt5-default. Regardless, nearly every qt5 package is already available on the system, which should provide all necessary functionality for the package. See below.

```bash
libqt5charts5/kali-rolling,now 5.15.15-2 amd64 [installed,automatic]
libqt5charts5-dev/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5concurrent5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5core5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5ct-common1.8/kali-rolling,now 1.8-2+b1 amd64 [installed,automatic]
libqt5dbus5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5designer5/kali-rolling,now 5.15.15-6 amd64 [installed,automatic]
libqt5designercomponents5/kali-rolling,now 5.15.15-6 amd64 [installed,automatic]
libqt5gamepad5/kali-rolling,now 5.15.15-2 amd64 [installed,automatic]
libqt5gamepad5-dev/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5gui5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5help5/kali-rolling,now 5.15.15-6 amd64 [installed,automatic]
libqt5location5/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
libqt5location5-plugin-mapboxgl/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
libqt5location5-plugins/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
libqt5multimedia5/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5multimedia5-plugins/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5multimediagsttools5/kali-rolling,now 5.15.15-2 amd64 [installed,automatic]
libqt5multimediaquick5/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5multimediawidgets5/kali-rolling,now 5.15.15-2 amd64 [installed]
libqt5network5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5networkauth5/kali-rolling,now 5.15.15-3 amd64 [installed,automatic]
libqt5opengl5-dev/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed]
libqt5opengl5t64/kali-rolling,now 5.15.15+dfsg-5 amd64 [installed,automatic]
libqt5pas1/kali-rolling,now 3.8+dfsg1-1 amd64 [installed,automatic]
libqt5positioning5/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
libqt5positioning5-plugins/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
libqt5positioningquick5/kali-rolling,now 5.15.15+dfsg-3+b1 amd64 [installed,automatic]
[... 69 lines cut ...]
qt5ct/kali-rolling,now 1.8-2+b1 amd64 [installed]
qt5dxcb-plugin/kali-rolling,now 5.7.12-1 amd64 [installed]
```

So, use of `deb` package abandoned.

### Install with AppImg

Installation with an AppImg file caused the following error:

```bash
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/USER/Applications/EspoTek_Labrador-65a3ad2-x86_64_3d43df5ece7e8a4e449b54d56e93fc3f.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/USER/Applications/EspoTek_Labrador-65a3ad2-x86_64_3d43df5ece7e8a4e449b54d56e93fc3f.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
fuse: memory allocation failed
```

Squashfs encountered an error with the AppImg compression. 

## Summary

Was finally able to get it to build from source on Linux, but not sure how useful this will be on a headless box.

turboencabulator, 2025-04-19:
For the `struct timeval tv` error:  I think the file containing this line is missing an include to pick up the `struct timeval` definition.  Not sure why we haven't seen that error on other Unix-like OSes.  I've just pushed a change to move this code to a different file, let us know if this allows you to get any further.

I'll look into the clang/openmp issue, I was seeing something similar while trying to build with clang on Linux.

The Debian package has several problems I'm slowly working through, one of the biggest is that the dependencies aren't correct.  I believe `qt5-default` was a precursor to the `qtchooser` package, that both no longer exists and isn't something this package actually depends on.

turboencabulator, 2025-04-19:
On Ubuntu I was able to build with clang (using `qmake -spec linux-clang`) only after installing the `libomp-dev` package.  I'm not familiar with OpenBSD, but maybe it might have a similar `libomp` package that would provide the missing header?

anoduck, 2025-04-21:
@turboencabulator I regret not getting back to this sooner, since you were so prompt to respond. There is currently no rush towards any resolution, as I have yet to figure out how I would like to patch my inherited oscilloscope probe into the board. 

I thank you for bringing clarity to exactly which "omp" is being referenced. It appears only parts of OpenMP are currently implemented in OpenBSD, libomp not being one of them. Seven years ago, a [port of OpenMP](https://github.com/jasperla/openbsd-wip/tree/master/devel/openmp) was created in the "WIP" ports tree repository. It clearly has not been touched since, and fails to build. For which, there is not much we can do about. 

The important thing is it built on Linux. Truth be told, I have been playing around with the idea of purchasing a pineboard to accompany the labrador for greater portability.

The purpose of opening the issue was to notify the developer about difficulties encountered. I think we have covered this.

closed 2025-04-21 by anoduck