espotek-org/Labrador #338: Strange readings from function generator

AlphaDragon601, 2025-07-15:
---
Strange and garbled waves from the function generator
---

### PROBLEM DESCRIPTION
For reasons I didn't spot I'm suddenly getting garbled waves from the function generator. Both oscilloscope channels show it even when plugged directly into the board. I don't have another function generator handy but any signals I put in by hand (waggling a potentiometer) doesn't seem to have the issue. I've recalibrated the osc. several times with no change. Hoping its not a hardware issue but not impossible.

### REQUESTED INFORMATION
- [x] Searched the problem in [issues](https://github.com/espotek-org/Labrador/issues)
- [x] Searched the problem in [discussions](https://github.com/espotek-org/Labrador/discussions)
- [x] Searched the problem in the [docs](https://github.com/espotek-org/Labrador/wiki)

### TO REPRODUCE
Not sure, possible contact to the board? I'm using a separate 9v power supply with this project so thats the most any stray wire could apply. 

### EXPECTED BEHAVIOUR
A clean sine wave (or other wave) output from the function gen.

### SCREENSHOTS
at 3hz 1V sine:

[image]

at 10hz 1v:

[image]

at 50hz 1V

[image]

at 100hz 1V

[image]

EspoTek, 2025-07-15:
Is this what you see when directly wiring the function generator to the scope input, without any external circuitry attached?

If so can you shoot through a screenshot of the app as well as a quick photo of your setup?

AlphaDragon601, 2025-07-16:
[image]

[image]

my setup and the wiring that I used for the previous screenshots. The board is connected to my laptop running opensuse with the x86 appimage.

[image]

results from turning on doubled sample rate at 50hz^

EspoTek, 2025-07-16:
OK, that narrows it down a bit!  Long story short, the USB packets are, for whatever reason, being received by libusb out of order and so the waveforms are being reconstructed incorrectly by the desktop software.

Are you running the device through a USB hub?  Try changing to a direct connection and it should go away.  If you're directly connected, try switching to a hub (if you've got one lying around).

If your laptop has both USB 3.0 and 2.0 ports, try connecting the device directly to both of them too.  I had a PC once where the Labrador worked fine on 2.0 but not 3.0.

AlphaDragon601, 2025-07-16:
I decided to update my packages and labrador installation. In the mean time I tested on my windows PC which had no issue. When I came back to my laptop and tried again it activated firmware flashing for some reason (the new labrador install?) and now things seem to be working again. I'm assuming the firmware flashing was what fixed it.

EspoTek, 2025-07-16:
When you plug it into Windows it loads on a different firmware optimised for libusbK.  Connecting it back to Linux will load on whatever firmware was running before.

Glad to hear it's working now, but it probably wasn't the flashing that did it!

Either way, I'll mark this as solved for now.

closed 2025-07-16 by EspoTek